envy/esp-knx-ip #3: Phy and group adresses

pit001, 2017-08-21:
Hi. I have checked your code and example - great working ! Thank you. 
One small thing: Please use the standard syntax fpr physical adresses with pots like 0.0.1 in your webpage. Group adresses with slashes like 0/0/1. It makes things more clear :-)

For finalization it would be great to seperate the webserver implementation and the KNX finctionality.

best regards, Peter

envy, 2017-08-22:
Hi,

> One small thing: Please use the standard syntax fpr physical adresses with pots like 0.0.1 in your webpage. Group adresses with slashes like 0/0/1. It makes things more clear :-)

Will change this with one of the next commits.

> For finalization it would be great to seperate the webserver implementation and the KNX finctionality.

I don't know how I want to handle this, yet. My plan was to be able to flash the same firmware onto multiple devices and then configure the GAs in some other way so that they are not hardcoded.
My first idea was an application running on a PC sending special UDP packets that would configure the device, but this is too complex. So self-contained configuration via webserver was easiest. I'll probably make it disableable during compile time with a define or something.
Maybe one day I can trick the ETS into programming ESPs ;D

pit001, 2017-08-22:
Hi envy !
To ctrl and setup the config via webpage is a good way - only to have the webserver implementation and the KNX object in the same files makes it not easy to use it in other projects  - i.e if there already a webserver is running.
my 5ct :-)
In my case a #undefine it - no problem. 
regards, Peter

envy, 2017-11-12:
I changed the notation of physical address in the sebserver to dots. I also added the possibility to specify own webserver object, so that the knx system reuses the same webserver as other code. However, I need to add the possibility to specify a prefix for the knx endpoints as to not clash with other endpoints.

closed 2017-11-12 by envy