envy/esp-knx-ip #4: More examples please

mrWheel, 2017-11-12:
Hi Nico,

Could you please add some more examples?

I have trouble figuring out where param’s come from and where they go to ....

Thanks!!!

envy, 2017-11-12:
Hi, 

I don't quite understand your question. What parameters are you referring to?
I added a simple environment sensor example that might help you getting started.

Cheers, Nico

mrWheel, 2017-11-12:
Hi Nico,

Thanks for your prompt reply!

I’m not sure myself ...

My friend recently bought a house that is totally KNX wired.
I installed KNXD on a Raspberry and that works great as a KNX to ethernet gateway. Also installed Homebridge so he can switch lights with iOS devices.
But it’s a steep learning curve understanding KNX and exploring the possibility's. 

What’s nice off your esp-knx-ip implementation is the possibility to create cheap (wireless) KNX devices (switches and sensors).

I just don’t understand how the callback function works. Is the example the same as used in the video with the intensity changing LED?

Maybe I should first use the example and see what happens...

Are you still developing the library?
Great work!

Regards,

Willem Aandewiel
The Netherlands
gsm: 0653 - 499 551
www: http://willem.aandewiel.nl/
---
Thank God I'm Atheist

Verstuurd vanaf mijn iPad

envy, 2017-11-12:
I don't know which video you are referring to.
Basically it works like this:
1. You register a callback function with `register_callback`.
2. Using the integrated webserver, you assign a group address to the callback.
3. Everytime a knx telegram is received, we loop through all registered callbacks and check if the recevier group address matches. If yes, we call that function.

This allows you to flash the same firmware onto multiple devices and then configure them differently over wifi. No hardcoded addresses are used.

Inside the callback you get access to the payload and the method, i.e., the type of telegram (write, read, answer) and can act accordingly.

> Are you still developing the library?

Yes, every now and then I update something. Currently I don't have a KNX installation therefore development is a bit stalled, however this will change at some point next year.

mrWheel, 2017-11-12:
My mistake,

I was referring to this video “https://youtu.be/CtPy1ornSok” .. but thats an other author. I mixed things. Sorry.

But thanks for the sensor example and the explanation. I will try it out tomorrow!

Regards,

Willem 
>

mrWheel, 2017-11-13:
Nico,

This is a great library!!!

Still some questions ...

I have programmed two ESP's 
- one as a sender (sends a telegram every 10 seconds)
- one as a receiver

If I configure the same callback address and the same GA address the Receiver receives (most) of the packages send from te sender.

But not all .. Is there a way to transmit an "acknowledge" telegram to the sender so that he knows the telegram was received and processed and if there is no acknowledgement he can transmit the same telegram again?

An example of how to implement that (on both sides) would be fantastic! 

As a side note: I realise those questions are part due to my little understanding of KNX (but it's growing!).

Thanks in advance!

envy, 2017-11-14:
> If I configure the same callback address and the same GA address the Receiver receives (most) of the packages send from te sender.

Yeah, I had similar problems if I send too many telegrams in a short timespan. Sometimes the receiver would not get them, if it's currently handling a previously received telegram. I found that disabling debug serial output helped.

> But not all .. Is there a way to transmit an "acknowledge" telegram to the sender so that he knows the telegram was received and processed and if there is no acknowledgement he can transmit the same telegram again?

That is a good question. I'm not an expert so my understanding of KNX is not that good but as far as I know telegrams are not acknowledged. There is a ACK type for telegrams, but I don't know how/when to use it. From the parts of the specification I've read, it seems to not be used for normal communication. But I could be wrong.

mrWheel, 2017-11-15:
> Yeah, I had similar problems if I send too many telegrams in a short timespan. Sometimes the receiver would not get them, if it's currently handling a previously received telegram. I found that disabling debug serial output helped.

This can hardly be the case as I only send 1 telegram every 10 seconds ... Even if the debug serial output is disabled ..

envy, 2017-11-15:
Hm, then I sadly don't know what happens. Do you have a delay() in your loop() or are you doing any other things in loop()? I sadly don't know what happens inside the ESP arduino framework when a UDP packet arrives. I found that sending from an ESP works quite reliable but receiving is sometimes weird. Maybe I'm missing some config options.

sisamiwe, 2018-02-18:
Hi,
there was the request for more examples.

Here is a forum having some examples already done.
@envy 
Do you think there is a change to bring them to your repository?