envy/esp-knx-ip #5: Further instructions on how to connect with existing KNX system

mishoboss, 2017-11-13:
Hi, this project looks like a great way to add inexpensive custom devices to a KNX system. But I miss some key information on how exactly to do that. I see it uses 224.0.23.12:3671 UDP, so the ESP device acts as a coupler or something? For example I have a Weinzierl 730 ethernet gateway. How is supposed ESP device to be connected to the KNX system? And what is the role of ETS in that case?

Thanks in advance! :)

envy, 2017-11-14:
Hi,

the library implements KNXnet/IP, so you can build KNX IP devices with it. You need a KNX IP Router that provides the connection between TP and IP. Your gateway only supports tunneling, so it does not send out telegrams via multicast, which is what the library uses. The Weinzierl 751 is a router, for example.
You could also try to build your own router by installing [knxd](https://github.com/knxd/knxd) and using that as a router.

The ETS only comes into play for configuring the filter table on the router. For my ESP I simply added a dummy device to the IP line in my ETS project so that it knows which TP devices talk to my IP devices so it can create the correct filter table for the router. The filter table is needed, so that only telegrams needed on both sides are passed through the router, it basically acts like a firewall. The ETS is not used to configure the ESPs, this is done via a webbrowser.

pixilarion, 2018-02-13:
First of all, congrats on the work already done! I seem to have the same problem as the original poster: my IP gateway only supports tunnel connections. @envy , are you planning on adding support for this in the near future?

envy, 2018-02-13:
Thanks! No, support for tunneling is not planned. Gateways only support a very limited number of concurrent tunnels, e.g., five. So being able to have only five devices is kinda against the point. 

As said above, I should be able to build a router with knxd. Connect knxd via tunneling to your gateway and let it serve the router-part. knxd will then take care of the multicast communication and forward it to the gateway. I don't know the exact configuration flags for knxd, but I think that this is possible.

pixilarion, 2018-02-13:
True, tunneling connections are limited. In my case it is more than enough: OpenHAB and ETS only use 2 connections.  Since I'm trying to build only one self-contained "KNX to 433Mhz gateway" a tunneling connection would suffice. So, I think, for the time being, I will be soldiering on with my attempt to implement the tunneling protocol. Although, your knxd suggestions looks like an easier way, but would rely upon another device to build my "KNX to 433Mhz gateway" which I was aiming to avoid.

closed 2020-03-06 by envy