envy/esp-knx-ip #10: What IDE are you using?

olexabondarenko, 2017-11-28:
Greetings, Envy. Tell me please what IDE you use. I modified the example file for the DHT22 sensor in Arduino IDE a little. I would like to work in some other IDE

envy, 2017-11-28:
Hi,

For the sketches I use the Arduino IDE. The library itself was written with Sublime Text. I don't know how to use the Arduino stuff from some other IDE.

Cheers, Nico

olexabondarenko, 2017-11-28:
Thanks for the answer. Now I'm trying to cope with the eclipse =) If interested, I'll post a sketch with DNT22

olexabondarenko, 2017-11-29:
Thanks for the answer. Now I'm trying to cope with the eclipse =) If interested, I'll post a sketch with DHT22

envy, 2017-11-29:
Sure, more examples are welcome.

olexabondarenko, 2017-11-30:
Greetings, Nico. I added the sample file for DHT22 to the repository. But I'm not sure whether he appeared. With Git, I'm just trying to figure it out.

Still there is a question, "void write2ByteFloat" sends 3 bytes and in "ETS4" the incorrect data come, though "OpenHab2" accepts correctly. does not read last byte

ascillato, 2018-03-09:
Seems to be similar to #33, the checksum.

@olexabondarenko try to change the line 69 on [esp-knx-ip-send.cpp](https://github.com/envy/esp-knx-ip/blob/master/esp-knx-ip-send.cpp#L69) for `udp.write(buf, len-1);`

In my case this works with to use ESP-KNX-IP with Home Assistant

closed 2020-03-06 by envy