envy/esp-knx-ip #11: Step Up/Down Dimming 3.005/3.007, 4 bit

mrWheel, 2017-11-28:
Hi Nico,

Is there a way to send a 4bit telegram? For StepUp/Down dimming needs this but I can't see how..

[image]

... or am I missing something (again!)??

Pressing the KNX switch to dimUp/Down the lamp the monitor shows:
```
LPDU: BC 11 84 10 0E E1 00 89 A0 :L_Data low from 1.1.132 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 09 
LPDU: BC 11 85 10 0E E1 00 88 A0 :L_Data low from 1.1.133 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 08  
```
Writing 07, 08 or 09 values using writeBit() I get this:
```
LPDU: BC 11 C8 10 0E E1 00 81 E4 :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 01 
LPDU: BC 11 C8 10 0E E1 00 81 E4 :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 01 
```
Using write1ByteInt() the monitor shows:
```
LPDU: BC 11 C8 10 0E E2 00 80 09 EF :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write 09 
LPDU: BC 11 C8 10 0E E2 00 80 08 EE :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write 08 
```

mrWheel, 2017-11-28:
I added this function:
```
void ESPKNXIP::send4Bit(address_t const &receiver, knx_command_type_t ct, uint8_t nible)
{
  uint8_t buf[] = {(uint8_t)nible};
  send(receiver, ct, 1, buf);
}
```
and now I can send telegrams with the right payload. The KNXD monitor shows:
```
LPDU: BC 11 C8 10 0E E1 00 87 E2 :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 07 
LPDU: BC 11 C8 10 0E E1 00 88 ED :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 08 
LPDU: BC 11 C8 10 0E E1 00 89 EC :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 09 
LPDU: BC 11 C8 10 0E E1 00 88 ED :L_Data low from 1.1.200 to 2/0/14 hops: 06 T_DATA_XXX_REQ A_GroupValue_Write (small) 08
``` 
Which is what it should be. Only a check for the values 0x00 to 0x0F should be added to the function (I think)...

If you approve, will you please add this function to this already great library?

envy, 2017-11-28:
Hi,

I probably forgot to add this, I will when I get home in a few hours.

Can you send me a link to the documentation you are using? I have the official spec at home but I doesn't look like what you showed.

Cheers, Nico

mrWheel, 2017-11-28:
http://knx.com.ua/attachments/article/132/KNX-basic_course_full.pdf

Willem 

gsm: 0653 - 499 551
www: http://willem.aandewiel.nl/

---
Thank God I'm Atheist

Verstuurd vanaf mijn iPad

envy, 2017-11-28:
Done. I also added a version for 2 bits, as the spec also specifies this.

closed 2017-11-28 by envy

mrWheel, 2017-11-28:
Hi Nico,

This is great!
And lighting fast!

Regards,

Willem

commit eb22a150 (2017-11-28): Renamed senBit to send1Bit as well as write/answer functions

Added missing functions for sending 2 and 4 bit values as well as
write/answer versions.

This fixed #11