envy/esp-knx-ip #13: example environment-sensor does not compile

git-olicat, 2018-02-13:
error message:

-----------------------------------------------------------------------------------------------------------
esp_knx_environment_sensor\esp_knx_environment_sensor.ino: In function 'void setup()':

esp_knx_environment_sensor:37: error: invalid conversion from 'void (*)(knx_command_type_t, const address_t&, uint8_t, uint8_t*) {aka void (*)(__knx_command_type, const __address&, unsigned char, unsigned char*)}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]

   knx.register_callback("Read Temperature", temp_cb);

                                                    ^

In file included from esp_knx_environment_sensor\esp_knx_environment_sensor.ino:7:0:

C:\Users\engel\Documents\Arduino\libraries\esp-knx-ip-master/esp-knx-ip.h:315:19: error:   initializing argument 2 of 'callback_id_t ESPKNXIP::register_callback(String, callback_fptr_t, void*, enable_condition_t)' [-fpermissive]

     callback_id_t register_callback(String name, callback_fptr_t cb, void *arg = nullptr, enable_condition_t cond = nullptr);

                   ^

esp_knx_environment_sensor:38: error: invalid conversion from 'void (*)(knx_command_type_t, const address_t&, uint8_t, uint8_t*) {aka void (*)(__knx_command_type, const __address&, unsigned char, unsigned char*)}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]
-----------------------------------------------------------------------------------------------------------

Bu I need this example to adjust my own source to the new library.
Could you please fix or explain?
Thank you!

Regards, Oliver

envy, 2018-02-13:
Hi,

can you check with the latest commit? I fixed it, but can't test right now if there are any other errors.

git-olicat, 2018-02-13:
fixed!
Thank you!

Regards, Oliver

closed 2018-02-13 by git-olicat