envy/esp-knx-ip #14: SONOFF example also with additional sensors?

sisamiwe, 2018-02-18:
Hi,
first to say this is a great project!

So nevertheless I'm wondering whether it is (already) possible to add additional sensors like a DS18B20 to the GPIOs of the SONOFF and get there measurements also via knx?
This is a nice add-on the Tasmota firmware (MQTT) already has.

Thanks for the response.

envy, 2018-02-28:
Hi,

> first to say this is a great project!

Thanks!

> So nevertheless I'm wondering whether it is (already) possible to add additional sensors like a DS18B20 to the GPIOs of the SONOFF and get there measurements also via knx?
This is a nice add-on the Tasmota firmware (MQTT) already has.

I don't know yet, I don't want to recreate everything Tasmota has done.
I don't want to offer a finished solution for every imaginable use-case but rather a library that can be used to build specialized firmwares.
The examples are there to show the feature-set of the library in a minimal setting.

ascillato, 2018-05-14:
For anyone who is interesed, now [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) supports KNX thanks to this great library.

Also, there are compiled binaries [here](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases)