envy/esp-knx-ip #15: Webserver integration

ggs67, 2018-02-24:
Please make the webserver integrable with an existing web server of the main app, bei passing the existing server and an absolute path to the library at start and builc all internal path relative to th egiven one.

envy, 2018-02-24:
Hi,

this is already possible. Call `knx.start()` with a pointer to a `ESP8266WebServer` (see [here](https://github.com/envy/esp-knx-ip/blob/master/esp-knx-ip.h#L336)) and change `ROOT_PREFIX` to a path you want (see [here](https://github.com/envy/esp-knx-ip/blob/master/esp-knx-ip.h#L29)).

closed 2018-02-24 by envy

ggs67, 2018-02-24:
Well, that does not allow any alternat root path and does not work if root is already used, which usually is the case. With start(server) on e has to edit the esp-knx-ip.h header and modify ROOT_PREFIX to something likse "/knx" then it works. 

so start should get another parameter allowing to specify the path below which the knx webpages are to be found.

Without this prefix, the implementation is also vey collision prone, for instanmce if the application already has a /config webpage.

envy, 2018-02-24:
>Well, that does not allow any alternat root path and does not work if root is already used, which usually is the case.

Yes it does, just set `ROOT_PREFIX` to your desired path, e.g., `/knx`, if you use `/` for your own application.

> With start(server) on e has to edit the esp-knx-ip.h header and modify `ROOT_PREFIX` to something likse "/knx" then it works.

This is how it's supposed to work. You choose a path and set `ROOT_PREFIX` to that. This way, all the strings are fixed during compile time instead of being dynamically build during runtime.

> so start should get another parameter allowing to specify the path below which the knx webpages are to be found.

No, that's the job of `ROOT_PREFIX`.

> Without this prefix, the implementation is also vey collision prone, for instanmce if the application already has a /config webpage.

Yes, that's why `ROOT_PREFIX` exists.

I see no advantage of moving this parameter to `start` and therefore making all the string concatenations runtime operations.

ggs67, 2018-02-27:
Well the advantage is obviously that th elkibarry needs no editing for each separate project. If one has a webserver and trhe other not. Also it would allow to have a backlink to the rrot on the KNX page if the prefix is given.

If you absolutely want to stick with the preprocessor macro, I would suggest the following change:

make #define ROOT_PREFIX commented out in place where now defined with a comment that it needs to be uncommented for change.

Just before the use of the root prefix for the other paths  then:

#ifndef ROOT_PREFIX
#define ROOT_PREFIX "" // DO NOT CHANGE this line, uncomment and change above !!!!
#define ROOT_PREFIX_USED 0
#else 
#define ROOT_PREFIX_USED 1
#endif 

an then use ROOT_PREFIX_USED to add a link back to the root on teh KNX page. Or you could of course simply strlen the prefix ;-)