envy/esp-knx-ip #17: KNX-IP Lite Version

ascillato, 2018-02-26:
Hi @envy,

Great Library!!! I was able to connect it to Home Assistant very fast and It is working great!!! :+1: Thanks

I have a feature request.

Can be modified a little your library so as to be just the KNX-IP Interface without the Webserver? Like an option to turn off the webserver and eeprom features, and to set manually the address and callbacks?
This request is because for using your library in a project that has its own webserver, the integration is very difficult.

If you can have the webserver and eeprom as optional, it can be easily integrated on ESPEasy, Sonoff-Tasmota, Espurna, etc.

What do you think?

envy, 2018-02-26:
Hi,

> Can be modified a little your library so as to be just the KNX-IP Interface without the Webserver?

Yes, I can add this.

> Like an option to turn off the webserver and eeprom features, and to set manually the address and callbacks?

Without EEPROM should already be possible, just don't call `knx.load()`.
EEPROM is only accessed there and by the webserver code.
However, there is no function yet for assigning callbacks from code, I will add this.

envy, 2018-02-26:
So I made it possible for `start()` to accept a `nullptr` which disables the webserver part.
I also added a `callback_assign` function to assign grozp addresses to callbacks. Together with the `config_set_type` functions that should be enough to configure the library without using the web UI.

You can now achieve a webserver-less and EEPROM-less library by calling `knx.start(nullptr)` instead of `knx.start()` and removing `knx.load()`. Configuration is then done using `callback_assign` and `config_set_type` or the default value of config items.

Please test the changes.

ascillato, 2018-02-26:
wow!!! so fast!!! I will test it. Thanks a lot :+1:

ascillato, 2018-02-28:
Hi, works great! :+1: 

I made small changes for make it work.

Added few functions to the library:

```
// Physical Address Set and Get:
config_set_physical_addr ( area, line, member );
config_get_physical_addr ();

// The Reverse of GA_to_address ( area, line, member ):
area_from_address ( address_t );
line_from_address ( address_t );
member_from_address ( address_t );

// Address set by area, line, member on GA and Callbacks>
config_set_ga ( id, area, line, member );
callback_assign ( id, area, line, member );
```

Also, I made a _Sonoff_DHT11.ino_ as example of manual set of address on the code. With some comments on using the `Start()` function

I'm making a Pull Request with this changes for you to review.

ascillato, 2018-02-28:
The `Start(server)` function to use an already running webserver is a very good idea also with `root_prefix` :+1: 

An idea: 

When Root_Prefix is different than "", a Home or Return Button should show up on the KNX Config Page. What do you think?

envy, 2018-02-28:
Sorry, but I'm going to reject all those pull requests.

#23 You added a bunch of addresses specific to your use-case and there is also a DHT sensor added. The example is meant to show the feature-set of the library in a minimal way.
#22 I don't want to move the files into a subfolder. AFAIK it is fine for Arduino libraries to have the files in the root folder
#21 See above.
#20, #19, #18 I added the physical address functions myself, but named them differently (17c390d8ae8d71b1bc7e99226c2dea21f8ddf7c6).
I also extended the `address_t` union to include `ga` and `pa` members making it possible to read out area, line and member easily without extra functions.
Lastly, I want the public API to expect `address_t` and not have overloads for every function with `uint8_t area, uint8_t line, uint8_t member`. That's why `GA_to_address` exists, to make this easier. But now, you can even do
```c++
address_t a;
a.ga.area = 1;
a.ga.line = 2;
a.ga.member = 3;
```
or `address_t a = {.ga={line, area, member}}` (Yes, you need to switch line and area) (432439a69c518d4eb46d5c1e84bdd66dacb23fd9)

ascillato, 2018-03-02:
Hi,

No problem with those PR. I make those pull requests just as ideas to be discussed and as an example of setting addresses by code :+1: 
I like your approach for the tranformation of GroupAddress to area, line and member. It is more elegant.
I will re make my tests with your modifications.

As the ESP_KNX_IP library is intended to be a library to be included on other proyects, what do you think of:

1- Posibility of adding buttons with links (as you said) and buttons to callbacks?
2- Posibility of turning off the buttons EEPROM SAVE, EEPROM RESTORE and REBOOT?
3- Posibility of changing the webpage title?
4- Posibility of setting the webpage text (so as a language translation be posible by code)
5- Posibility of a custom arrange of all the objects on the webpage. Then, I will post here an example.

If you want, I can help you with the code and with the Wiki.

Thanks a lot.

ascillato, 2018-03-07:
Made the PR #26 #27 #28 to:

* Added #ifndef ROOT_PREFIX to have the option to set the Root Prefix before calling the #include <esp-knx-ip.h>, so as to eliminate the need of modifying the library.

* Changing the name of ntohs to __ntohs to eliminate conflict when using ESP8266 LWIP library.

envy, 2018-03-08:
Hi, I created some issues to track some ideas seperately. (#36, #37, #38)

> Posibility of adding buttons with links (as you said) and buttons to callbacks?

What do you mean with "buttons to callbacks"? Buttons to issue callbacks from the webui?

> Posibility of a custom arrange of all the objects on the webpage. Then, I will post here an example.

How would you specify arrangement of objects? xy-coordinates? Some kind of grid?

ascillato, 2018-03-09:
> What do you mean with "buttons to callbacks"? Buttons to issue callbacks from the webui?

Yes! exactly :+1: 

> How would you specify arrangement of objects? xy-coordinates? Some kind of grid?

Can be in the order you register them on the main application code. Also can be everything centered.
#36

ascillato, 2018-03-09:
Closing this issue as all the questions I had were answer. Thanks :+1: 

New topics continue in their specific issues.

closed 2018-03-09 by ascillato