envy/esp-knx-ip #24: WifiUDP.h cannot be found

McOrvas, 2018-03-04:
Hi envy,

Thanks for this very cool library! I have a short question about the included "WifiUDP.h" in "esp-knx-ip.h". On my system this header file is written "WiFiUdp.h", and so I have to change the notation of this to compile my project. Do you use a system which is not case sensitive or do I just have another version installed? I'm using "esp8266" in version 2.4.0.

Thanks and regards,
McOrvas

envy, 2018-03-04:
Hi,

Yeah, I use Windows... I copied this from some example and never bothered to check as it worked...

I fixed it, thank you.

closed 2018-03-04 by envy

McOrvas, 2018-03-09:
Sorry to write again about this issue, but the correct notation is "WiFiUdp.h" with an upper case "F".

commit f7f6bf05 (2018-03-04): Fixed #24 wrong name of WiFiUdp.h