envy/esp-knx-ip #33: Using ESP-KNX-IP Library to connect to Home Assistant

ascillato, 2018-03-08:
This is a workaround explanation for using **ESP-KNX-IP** Library to connect to **Home Assistant**.

The link between [Home Assistant](https://github.com/home-assistant/home-assistant) and a **KNX Router**, is the [XKNX Python Library](https://github.com/XKNX/xknx).

> **Side note:**
> If you don't have a **KNX Router**, you can use [KNXd](https://github.com/knxd/knxd) on the same Raspberry Pi than Home Assistant.

There is a bug using XKNX Library on Home Assistant, that makes XKNX Library not understanding [ESP-KNX-IP](https://github.com/envy/esp-knx-ip) (Issue https://github.com/XKNX/xknx/issues/112)

While waiting for response on the XKNX issue, the checksum can be eliminated changing [esp-knx-ip-send.cpp#L69](https://github.com/envy/esp-knx-ip/blob/master/esp-knx-ip-send.cpp#L69) for `udp.write(buf, len-1)`

With this you can have a working **ESP-KNX-IP** Library for testing on **Home Assistant** until the bug is solved.

envy, 2018-03-08:
So the checksum is a very interesting thing. In the KNX spec pdfs I have here, there is no mention of it, at least I can't find it at the moment.

I recorded some messages with tcpdump and found out that openHAB does not include a checksum to its packets whereas my KNX IP Router (ABB IPR/S3.1.1) is adding one but sets it to 0x00.

Maybe it is not even needed? I can't find the reference I used that said that there is a checksum field but everything worked in my environment with real KNX devices so I thought it is correct.

Maybe we should find out if it really is part of the spec or not.

ascillato, 2018-03-09:
Me neither. I could not find anything about the checksum.

I'm waiting the response of the author of XKNX Library for Home Assistant but seems that there is no need for checksum. May be that the KNX Devices are ingnoring the last byte, thats why the KNX network continue working.

Better could be add the ACK response and resend if NACK is received.

closed 2018-04-01 by ascillato