envy/esp-knx-ip #41: warnings during compiling

ascillato, 2018-03-09:
Hi,

When compiling on _Arduino IDE_, there are some warnings.

The compile ends OK and the Software run without problems. But these warnings need to be addresed in any way?

_Arduino Build Console Output:_
```
C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_1bit(const address_t&, knx_command_type_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:75:23: warning: narrowing conversion of '(((int)bit) & 1)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {bit & 0b00000001};

                       ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_2bit(const address_t&, knx_command_type_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:81:26: warning: narrowing conversion of '(((int)twobit) & 3)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {twobit & 0b00000011};

                          ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_4bit(const address_t&, knx_command_type_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:87:27: warning: narrowing conversion of '(((int)fourbit) & 15)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {fourbit & 0b00001111};

                           ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_3byte_time(const address_t&, knx_command_type_t, uint8_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:134:50: warning: narrowing conversion of '(((((int)weekday) << 5) & 255) + (((int)hours) & 31))' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {0x00, (((weekday << 5) & 0xE0) + (hours & 0x1F)), minutes & 0x3F, seconds & 0x3F};

                                                  ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:134:77: warning: narrowing conversion of '(((int)minutes) & 63)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {0x00, (((weekday << 5) & 0xE0) + (hours & 0x1F)), minutes & 0x3F, seconds & 0x3F};

                                                                             ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:134:93: warning: narrowing conversion of '(((int)seconds) & 63)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {0x00, (((weekday << 5) & 0xE0) + (hours & 0x1F)), minutes & 0x3F, seconds & 0x3F};

                                                                                             ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send_3byte_date(const address_t&, knx_command_type_t, uint8_t, uint8_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:140:29: warning: narrowing conversion of '(((int)day) & 31)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {0x00, day & 0x1F, month & 0x0F, year};

                             ^

C:\Users\Usuario\Documents\Arduino\libraries\esp-knx-ip-master\esp-knx-ip-send.cpp:140:43: warning: narrowing conversion of '(((int)month) & 15)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]

  uint8_t buf[] = {0x00, day & 0x1F, month & 0x0F, year};

                                           ^
```

envy, 2018-03-10:
Which version of the IDE are you using? I don't see any warnings on my machine.

ascillato, 2018-03-10:
Arduino IDE 1.8.2 but on the IDE by default, showing the warnings are off
and you have to turn it on.

Those are just warnings, not errors. The compilation works but those are
things to take into account sometime.

closed 2018-03-10 by envy

ascillato, 2018-03-10:
Great!!! Tested and no more warnings!!!!

Thanks :+1:

commit 4d8bc76d (2018-03-10): Some casts to 'fix' warnings. FIxes #41