envy/esp-knx-ip #51: On AyncUDP branch, Callbacks don't work with knx.start(nullptr)

ascillato, 2018-04-07:
Hi,

On the AyncUDP branch, if I start ESP-KNX-IP with `knx.start(WebServer);` or with `knx.start();` the library works perfectly.

But if I use the non webserver version: `knx.start(nullptr);`, callbacks don't work. Nevertheless, in this mode I can send telegrams without problem but I can not receive any.

What can I do?

ascillato, 2018-04-08:
I made the PR #48 and #52 for making ESP-KNX-IP work for a [modification](https://github.com/ascillato/Sonoff-Tasmota_KNX) to Sonoff Tasmota to add KNX IP to its features.

What do you think of these Pull Requests?

envy, 2018-04-09:
Hi,

callbacks work for me, even without a webserver.

Also, your changes don't actually change anything? You just replaced the `server != nullptr` with a dedicated bool and moved the website registrations to another function. The UDP part at the end of `__start` is called with and without your patch and that is reponsible for registering a packet handler function.

Did you check with debugging enabled that really no packet arrives?

envy, 2018-04-09:
I'm also using the async-udp branch with lwip 1.4 and as I said, I have no problems when passing nullptr for the webserver.

You can try adding a print into the lambda like so:

```c++
udp.onPacket([this](AsyncUDPPacket &packet) {
  DEBUG_PRINTLN("got packet");
  __loop_knx(packet);
});
```

to see if really no packet arrives.

ascillato, 2018-04-09:
Hi,

I found the bug! Was not a problem of ESP-KNX-IP.

When I started to add KNX to Tasmota I was using the webpage of the library. So the `knx.start(webserver);` was after the declaration of the Tasmota webserver.

Then, when change to use `knx.start(nullptr);` I moved it to an internal Tasmota function that calls all the drivers on device start. And that start is BEFORE Wifi. That's why the UPD listen was not working.

Now everything works. I ran several tests and goes smooth, So, I'm closing this issue and the PR #52.

Nico, thanks a lot for your time. It is very appreciated :+1:

closed 2018-04-09 by ascillato