envy/esp-knx-ip #53: Delete Functions needed

ascillato, 2018-04-09:
Hi @envy ,

When using and configuring the ESP-KNX-IP Library but only by code and not using the ESP-KNX-IP webpage, are needed new functions:

* callback_delete_assignment
* callback_delete_register

Can be added to the library?

Thanks a lot for your time! Great Library.

ascillato, 2018-04-09:
Made the PR #54 and #55 for adding this functions.

envy, 2018-04-09:
Deleting callback (and config and feedback) registrations that way this tricky, as it shifts the ids of callbacks that were registered after the to-be-deleted callback was registered.
An application would then need to shift (some) of their saved ids so that it matches again.
Example:
```c++
callback_id_t id1 = callback_register(...); // id1 = 0
callback_id_t id2 = callback_register(...); // id2 = 1
callback_id_t id3 = callback_register(...); // id3 = 2
callback_id_t id4 = callback_register(...); // id4 = 3
...
callback_delete_register(id2); // Moves the last two callbacks one slot up

// Now the callback that was id3 took the spot of id2 and id4 is now invalid
```

For assignments this is fine, as the application does not receive an id on assignment. So we can shift sstuff around. But having the `callback_delete_assignment` would take an argument, that the application does not even have as `callback_assign` returns `void` and not `callback_assignment_t`.

In the end, the structures could be modified to allow this, e.g., mark deleted spots as deleted and don't shift others. But then we would also need to reclaim them, skip deleted spots during iteration, etc.

I will add this.

ascillato, 2018-04-09:
Like your approach! :+1:

closed 2018-04-09 by envy

ascillato, 2018-04-10:
Tested and works really good !

Thanks !!!!! :+1:

commit 68522401 (2018-04-09): Added functions to delete callback assignments and registered callbacks.

callback_deregister and callback_unassign added.
callback_assign now returns a callback_assignment_id_t.

Internally, slots are marked as empty and merged when possible.

This can also be used for the config and feedback slots at a later point.

This fixes #53 and closes #54 and #55.