envy/esp-knx-ip #60: [ Question ] Sending and receiving a float type value

ascillato, 2018-06-12:
Hi Nico,

How are you?

I want to use your library to send and receive a float type value.

For sending a I use the `send_2byte_float` function that works great. It is being received by KNX without any problem.

Now I want to do the other way round. To receive a float type value. Do you have the code to transform the mantisa from the KNX DPT to a float value? When a float is received, the `msg.data_len` is 3, but the  `msg.data[0]` is always zero. So all the **2byte float** is in `msg.data[2]` and `msg.data[1]`.

So my question is if you have the fragment of code to decode `msg.data[2] `and `msg.data[1]` into a float variable ?

Thanks,

Adrian

envy, 2018-06-12:
Hi,

Yes, there is [ESPKNXIP::data_to_2byte_float](https://github.com/envy/esp-knx-ip/blob/master/esp-knx-ip-conversion.cpp#L38)

Cheers,
Nico

ascillato, 2018-06-12:
You Rock!

Thanks a lot :+1:

closed 2018-06-12 by ascillato