envy/esp-knx-ip #63: Static example does not compile

zbikmarc, 2018-11-21:
[Static example](https://github.com/envy/esp-knx-ip/blob/master/examples/static-config/static-config.ino) does not compile in Arduino IDE (`1.8.7`) with esp8266 board library version `2.3.0`. Same error for different boards selected.

Issue looks similar to what was in #13 
```
C:\SANITIZED\Arduino\esp_knx_static_example\esp_knx_static_example.ino: In function 'void setup()':

esp_knx_static_example:34:76: error: invalid conversion from 'callback_id_t {aka unsigned char}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]

   callback_id_t temp_cb = knx.callback_register("Read Temperature", temp_cb);

                                                                            ^

In file included from C:\SANITIZED\Arduino\esp_knx_static_example\esp_knx_static_example.ino:8:0:

C:\SANITIZED\Arduino\libraries\esp-knx-ip-master/esp-knx-ip.h:367:19: error:   initializing argument 2 of 'callback_id_t ESPKNXIP::callback_register(String, callback_fptr_t, void*, enable_condition_t)' [-fpermissive]

     callback_id_t callback_register(String name, callback_fptr_t cb, void *arg = nullptr, enable_condition_t cond = nullptr);

                   ^

esp_knx_static_example:35:71: error: invalid conversion from 'callback_id_t {aka unsigned char}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]

   callback_id_t hum_cb = knx.callback_register("Read Humidity", hum_cb);

                                                                       ^

In file included from C:\SANITIZED\Arduino\esp_knx_static_example\esp_knx_static_example.ino:8:0:

C:\SANITIZED\Arduino\libraries\esp-knx-ip-master/esp-knx-ip.h:367:19: error:   initializing argument 2 of 'callback_id_t ESPKNXIP::callback_register(String, callback_fptr_t, void*, enable_condition_t)' [-fpermissive]

     callback_id_t callback_register(String name, callback_fptr_t cb, void *arg = nullptr, enable_condition_t cond = nullptr);

                   ^

esp_knx_static_example:36:73: error: invalid conversion from 'callback_id_t {aka unsigned char}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]

   callback_id_t pres_cb = knx.callback_register("Read Pressure", pres_cb);

                                                                         ^

In file included from C:\SANITIZED\Arduino\esp_knx_static_example\esp_knx_static_example.ino:8:0:

C:\SANITIZED\Arduino\libraries\esp-knx-ip-master/esp-knx-ip.h:367:19: error:   initializing argument 2 of 'callback_id_t ESPKNXIP::callback_register(String, callback_fptr_t, void*, enable_condition_t)' [-fpermissive]

     callback_id_t callback_register(String name, callback_fptr_t cb, void *arg = nullptr, enable_condition_t cond = nullptr);

                   ^

exit status 1
invalid conversion from 'callback_id_t {aka unsigned char}' to 'callback_fptr_t {aka void (*)(const __message&, void*)}' [-fpermissive]
```

zbikmarc, 2018-11-26:
I created PR which should fix compilation problem and hopefully will not break callback functionality.

Looks like there is naming collision between `callback_id_t ` and callback functions.

closed 2018-11-26 by envy

zbikmarc, 2018-11-29:
I am afraid #64 indeed fixed compilation error but it looks like callbacks are not working properly :(

zbikmarc, 2018-12-16:
@envy
Well, now I think it works but based on my experiments it looks like root cause of this problem is UDP multicast. In my case it is very unstable. Sometimes about 50% of packages are delivered, sometimes less than just few %. Of course It might be issue with my WiFi setup (unifi) but google tells nothing about this and I think I've tried everything I could to increase reliability :(

envy, 2018-12-16:
Yes, UDP multicast is not reliable and sadly KNX does not have a resend/ack mechanism specified.

I also have a unifi setup and no problems, but also not that much wifi traffic.

You can try the async-udp branch which has asynchronous handling of incoming requests.

zbikmarc, 2018-12-17:
Tried async-udp branch and maybe works a bit better  - when I do read request every 60sec I have about 50% success. And this is only traffic in my small knx lab.
If it works for you with same code and obviously same hardware, I am pretty sure that problem is related to my network settings/hardware not to code itself.

PR #64 (2018-11-26): Change name for registered callback id
This PR changes name of created `callback_id_t` entities which fixes compilation problem caused by name collision between `callback_id_t` entities and callback function names.

Fixes #63 

Disclaimer: 
Currently I can not check if callback functions work correctly with this change.