envy/esp-knx-ip #68: KNX Communication - uncalled UDP response

HomeAsASystem, 2019-01-29:
Have you ever experienced this issue ?

[https://github.com/arendst/Sonoff-Tasmota/issues/5053](url)

I've got the same behaviour not only with Tasmota but also with the esp-knx-ip example

envy, 2019-01-29:
Hi,

no I haven't. What I gather from your bug report is this:

1. You boot the ESP device.
2. Some communication happens on the KNX Multicast IP
3. The ESP responds with packets of length 0. This only happens if the ESP has not sent own KNX packets before.

Is this correct? I have to investigate this, but unfortunately I don't have access to my lab this week, so maybe next week at the earliest.

HomeAsASystem, 2019-01-29:
Hi,
My description was not very clear.
After booting up the device it responds to every KNX multicast telegram which is send from any device to the KNX multicast address 224.0.23.12:3671. 
As long as the device has not sent any KNX/IP message the response is targeted to 0.0.0.0:0, but once the device has sent a KNX/IP message, the UDP response is send to the KNX multicast address 224.0.23.12:3671.
So in summary, the device always responds to the multicast telegram, but the destination address of the response changes, once the device has sent an own KNX/IP telegram. I believe the root cause is somewhere in the TCP/IP or IGMP implementation of the ESP8266 stack or in a wrong configuration of the IGMP function. I removed the igmp_joingroup statement in the WifiUdp Class, then the device does not respond anymore to the multicast telegrams, but it also does not react anymore on the KNX/IP telegrams send to the device.

d-a-v, 2019-01-29:
> As long as the device has not sent any KNX/IP message the response is targeted to 0.0.0.0:0, but once the device has sent a KNX/IP message, the UDP response is send to the KNX multicast address 224.0.23.12:3671.

This is still a bit unclear,

- "device" is ESP ?
- if so, `KNX/IP message the response is targeted to 0.0.0.0:0` is sent by ESP,
- `the UDP response is send to`, "response" is from the ESP ?

(I am maintainer in the core, especially the lwIP layer)

HomeAsASystem, 2019-01-29:
Sorry for being not precise enough,
below is a wireshark trace of the issue
192.168.1.40 is my ESP8266 Device
192.168.1.99 is my KNX-IP Gateway
224.0.23.12 is the KNXnet multicast address

[image]

I restartet the ESP Device before line 4041. Then the IP Gateway sends KNXnetIP messages to the KNX Multicast address. The ESP device@192.168.1.40 responds with an UDP message with the destination 0.0.0.0.
At line 4083 the ESP device@192.168.1.40 sends its first KNXnetIP telegram to the KNX multicast address.
Then at line 4093 another KNXnetIP telegram is send by the KNX-IP Gateway. Then the EPS Device@192.168.1.40 is the responding not anymore to 0.0.0.0 but to the KNX multicast address 224.0.23.12.
This leads then to a warning in my OpenHAB server, as it detects invalid messages to the KNX mutlicast address.
[image]

hope this helps to understand the issue
br

envy, 2019-01-29:
Are the messages sent by your gateway destined for the ESP (e.g., reads or writes to the ESP)? Or are they destined for some other KNX device?

If it's the first, that means, the ESP is answering to the wrong address.
If it's the second, then the ESP is answering the messages, it's not supposed to answer to.

envy, 2019-01-29:
> the messages of the IP gateway have no specific target, as it is the nature of the KNXnetIP protocol that all communication is send to the KNX multicast group address. Also the ESP device sends all its KNX messages to the KNX multicast address.

I meant, if your gateway is sending a KNX group read/write to a group address that the ESP is configured to react upon, i.e. you send group write to 1/2/3 and the ESP toggles a lamp and then send back a status to 4/5/6. I want to know if this answer message is instead send to 0.0.0.0.

Or if your KNX message is sent to a group address that is not configured on the ESP and the ESP is answering anyway even though it should not do that.

HomeAsASystem, 2019-01-29:
the Gateway is sending all sort of GA's as my openHAB server reads all the KNX GA's on the KNX TP installation.
The ESP device responds to all messages of the IP Gateway. No matter if the GA's are addressed to the ESP device or not. The 192.168.1.40 device even has no receiving GA's configured as there is only a BME680 sensor connected to.

envy, 2019-01-29:
Ok, so it is sending packets, even though it should not. I have to investigate if this is a bug on my side or one inside the ESP framework (or lwip). But if the responding stops when the ESP sent a message on its own, then it sounds more like a bug inside the ESP framework (or lwip).

Did you try the different lwip versions available?

HomeAsASystem, 2019-01-29:
another missunderstanding. when the ESP sent a message on its own, it still continues to repsond to any incominc KNXnetIP multicast address, it just changes the destination IP of its reponse from 0.0.0.0 to 224.0.23.12 with an invalid UDP message. The ESP device is not supposed at all to respond to the KNXnetIP telegrams from the IP-KNX gateway

HomeAsASystem, 2019-01-29:
I also think it is related to the esp framework. I've disabled  any sending and receiving within your knx code. As soon as the esp device joins an igmp group this behaviour turns up

roelbroersma, 2019-10-01:
I am seeing the same behavior.

The ESP just always answers with an UDP 0  (zero) to the 0.0.0.0  IP address.    Very strange...

I never seen this before and I have seen a lot of KNX routers like Gira (all variants), MDT,..and more.

Even when this esp-knx-ip Library is only loaded and Nothing is configured..  the ESP just starts to 'answer'  all Multicast calls with the UDP 0 message.

ascillato, 2019-10-01:
Please, try latest arduino core from arduino. The STAGE version that will became soon the 2.6.0 has solved several bugs. I use it and I couldn't reproduce your issue.