envy/esp-knx-ip #70: Send String

Poseidon1982, 2019-07-08:
Hi Envy,

first of all thanks for this great work.
Would it be possible to add also a send string function?
I have some text information which I would like to send to the KNX bus.
Like the direction of the wind from a wind sensor.

Many thanks.
Poseidon

envy, 2019-07-09:
Hi,

I added a string sending method for DPT16. Please try it out, I did not test it.

Poseidon1982, 2019-07-09:
Many thanks Envy.

I tried to compile my sctach but it didn't work.
In the file "esp-knx-ip.h" I delete the parts "knx_command_type_t ct,".

So from:
void write_14byte_string(address_t const &receiver, knx_command_type_t ct, const char *val) { send_14byte_string(receiver, KNX_CT_WRITE, val); }

to:

void write_14byte_string(address_t const &receiver, const char *val) { send_14byte_string(receiver, KNX_CT_WRITE, val); }

Now it is compilingand tomorrow I will do a test with the sensor. I keep you posted.

Many thanks.
Poseidon

envy, 2019-07-09:
Whoops, sorry.

Copy and paste error on my side. I fixed this in the repo, too.

Poseidon1982, 2019-07-10:
Hi Envy,

I tested today and now it working but I had to do some changes in esp-knx-ip-send.cpp.
Following is the change:

void ESPKNXIP::send_14byte_string(address_t const &receiver, knx_command_type_t ct, const char *val)
{
	uint8_t buf[15] = {0x00};
	int len = strlen(val);
	if (len > 15)
	{
		len = 15;
	}
	memcpy(buf+1, val, len);
	send(receiver, ct, 15, buf);
}

I got some strange values and I assume that there was something in buf. So I took the value from the examples above the file {0x00. 
And the length have do be 15 otherwise ETS will not show only 13 bytes.
I don't really understand how it works so I did try and error and end up with a working result.

Many thanks
Poseidon

envy, 2019-07-11:
Thanks, fixed. I forgot to initialize the array and account for the first extra byte being needed.

closed 2019-07-11 by envy