envy/esp-knx-ip #73: Receive messages on group 2/1/0

brunohorta82, 2019-09-03:
Hello, thanks for the great library :)
I register my switches like this way:
` for (auto &item : items)
  {
    callback_id_t switchesCallbackId = knx.callback_register(String(item.name), switchesCallback, &item);
    knx.callback_assign(switchesCallbackId, knx.GA_to_address(2, 1, i));
    i++;`

the final result for 2 switches is like 2/1/1 and 2/1/2, I have any method to handle with the telegrams sent to 2/1/0 ?
if I receive on message on this group address I want to turn on/turn off all switches.

brunohorta82, 2019-09-03:
I made this way, is correct?
  callback_id_t switchesCallbackId = knx.callback_register("ALL", allwitchesCallback);
  knx.callback_assign(switchesCallbackId, knx.GA_to_address(2, 1, 0));

envy, 2019-09-05:
Hi,

Sure, you can do it this way.

closed 2020-03-06 by envy