envy/esp-knx-ip #75: Make MULTICAST_IP externally configurable using compiler switches

pfisterer, 2019-12-10:
I had to change `MULTICAST_IP` when using Tasmota's KNX support because a router frequently used in Germany (AVM's Fritz!Box) does not correctly implement multicast support. This, however, seems to be a [frequent issue with many routers](https://github.com/knxd/knxd/issues/16). 

As these faulty implementations forward Apple's Bonjour traffic, an option is to set the default multicast address to `224, 0, 0, 251`. Using Tasmota, this requires to modify esp-knx-ip in the `lib/` folder - which should be avoided.

Therefore, it could make sense to change `esp-knx-ip-0.5.2/esp-knx-ip.h` as follows:

Current version:
```
#define MULTICAST_IP              IPAddress(224, 0, 23, 12) // [Default IPAddress(224, 0, 23, 12)]
```

Updated version:
```
#ifndef
#define MULTICAST_IP              IPAddress(224, 0, 23, 12) // [Default IPAddress(224, 0, 23, 12)]
#endif
```

This would allow setting the MULTICAST_IP using compiler switches (e.g., `-DMULTICAST_IP=IPAddress(224, 0, 0, 251)`).

With this setting, I'm able to get a [working setup](https://github.com/knxd/knxd/issues/16#issuecomment-563967604) with KNX/IP to KNX/TP forwarding. So, I'm able to switch a MagicHome LED controller running Tasmota with a KNX/TP switch.

pfisterer, 2019-12-10:
@ascillato This is in the same project which we discussed yesterday in Tasmota's support chat (https://github.com/arendst/Tasmota/issues/7161#issuecomment-563392401). 

Maybe it would be helpful, to mention this in Tasmota's manual? It took me a while to find out that my wifi router was not correctly implementing multicast support. I've documented my workaround [here](https://github.com/knxd/knxd/issues/16#issuecomment-563967604).

ascillato, 2019-12-10:
@pfisterer In the meantime, you can make a PR to Tasmota in order to add that to the library.

ascillato, 2019-12-10:
ah, Did you test that approach? not always a key can be defined for a library

pfisterer, 2019-12-10:
Yes. To verify this setup, I've changed `lib/esp-knx-ip-0.5.2/esp-knx-ip.h` as follows: 
```
#ifndef MULTICAST_IP
#define MULTICAST_IP IPAddress(224, 0, 0, 251)  // [Default IPAddress(224, 0, 23, 12)]
#else
#warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
#endif
```
Afterwards, I've modified my `platformio_override.ini` of my Tasmota build and modified `build_flags` as follows: 

`build_flags = ${core_active.build_flags}  '-DMULTICAST_IP=IPAddress(224, 0, 0, 251)'`.

Compilation output (excerpt):
```
In file included from tasmota/tasmota_post.h:37:0,
                 from /Users/dennis/Documents/dev/Arduino/tasmota/Sonoff-Tasmota-MagicHome/tasmota/tasmota.ino:36:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
  ^

In file included from lib/esp-knx-ip-0.5.2/esp-knx-ip-config.cpp:7:0:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
  ^

Compiling .pioenvs/tasmota-knx/lib56f/esp-knx-ip-0.5.2/esp-knx-ip-send.cpp.o
In file included from lib/esp-knx-ip-0.5.2/esp-knx-ip-conversion.cpp:7:0:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
  ^
Compiling .pioenvs/tasmota-knx/lib56f/esp-knx-ip-0.5.2/esp-knx-ip-webserver.cpp.o
In file included from lib/esp-knx-ip-0.5.2/esp-knx-ip-send.cpp:7:0:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
  ^
Compiling .pioenvs/tasmota-knx/lib56f/esp-knx-ip-0.5.2/esp-knx-ip.cpp.o
In file included from lib/esp-knx-ip-0.5.2/esp-knx-ip-webserver.cpp:7:0:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
  ^
In file included from lib/esp-knx-ip-0.5.2/esp-knx-ip.cpp:7:0:
lib/esp-knx-ip-0.5.2/esp-knx-ip.h:39:2: warning: #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before [-Wcpp]
 #warning NOT USING DEFAULT SETTING, MULTICAST_IP has been defined before
```

ascillato, 2019-12-10:
> A PR would only affect the documentation

I meant, until @envy check this, we can modify the actual library in /lib folder in Tasmota repository for allowing this now.

ascillato, 2019-12-10:
Anyway, don't worry. I'm adding this now in the esp-knx-ip library in Tasmota.

envy, 2019-12-10:
Hi,

I added #ifndef guards around the IP and port.
I also had fun with multicast with my DLink switch. I had to completely disable IGMP snooping, otherwise it would not pass the KNX multicast traffic.

closed 2019-12-10 by envy

pfisterer, 2019-12-10:
Great, thank you guys!

commit eff93654 (2019-12-10): Make MULTICAST_IP and _PORT configurable from the outside. Fixes #75