envy/esp-knx-ip #77: Hello, has anyone managed to get this library to work with esp32?

robetson1989, 2020-03-21:
ATENCIÓN: la librería ESP-KNX-IP-Library_0.4_2894 pretende ejecutarse sobre arquitectura(s) esp8266 y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) esp32.
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send(const address_t&, knx_command_type_t, uint8_t, uint8_t*)':
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip-send.cpp:80:6: error: 'class WiFiUDP' has no member named 'beginPacketMulticast'
  udp.beginPacketMulticast(MULTICAST_IP, MULTICAST_PORT, WiFi.localIP());
      ^
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp: In member function 'void ESPKNXIP::__start()':
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp:98:67: error: no matching function for call to 'WiFiUDP::beginMulticast(IPAddress, IPAddress, int)'
   udp.beginMulticast(WiFi.localIP(),  MULTICAST_IP, MULTICAST_PORT);
                                                                   ^
In file included from /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFi.h:39:0,
                 from /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.h:57,
                 from /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp:7:
/Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiUdp.h:57:11: note: candidate: uint8_t WiFiUDP::beginMulticast(IPAddress, uint16_t)
   uint8_t beginMulticast(IPAddress a, uint16_t p);
           ^
/Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiUdp.h:57:11: note:   candidate expects 2 arguments, 3 provided
Se encontraron varias bibliotecas para "esp-knx-ip.h"
Usado: /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894
Se encontraron varias bibliotecas para "EEPROM.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/EEPROM
Se encontraron varias bibliotecas para "WiFi.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 No usado: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Se encontraron varias bibliotecas para "AsyncUDP.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/AsyncUDP
Se encontraron varias bibliotecas para "WebServer.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer
Se encontraron varias bibliotecas para "FS.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.
ATENCIÓN: la librería ESP-KNX-IP-Library_0.4_2894 pretende ejecutarse sobre arquitectura(s) esp8266 y puede ser incompatible con tu actual tarjeta la cual corre sobre arquitectura(s) esp32.
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip-send.cpp: In member function 'void ESPKNXIP::send(const address_t&, knx_command_type_t, uint8_t, uint8_t*)':
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip-send.cpp:80:6: error: 'class WiFiUDP' has no member named 'beginPacketMulticast'
  udp.beginPacketMulticast(MULTICAST_IP, MULTICAST_PORT, WiFi.localIP());
      ^
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp: In member function 'void ESPKNXIP::__start()':
/Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp:98:67: error: no matching function for call to 'WiFiUDP::beginMulticast(IPAddress, IPAddress, int)'
   udp.beginMulticast(WiFi.localIP(),  MULTICAST_IP, MULTICAST_PORT);
                                                                   ^
In file included from /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFi.h:39:0,
                 from /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.h:57,
                 from /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894/esp-knx-ip.cpp:7:
/Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiUdp.h:57:11: note: candidate: uint8_t WiFiUDP::beginMulticast(IPAddress, uint16_t)
   uint8_t beginMulticast(IPAddress a, uint16_t p);
           ^
/Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi/src/WiFiUdp.h:57:11: note:   candidate expects 2 arguments, 3 provided
Se encontraron varias bibliotecas para "esp-knx-ip.h"
Usado: /Users/robertson/Documents/Arduino/libraries/ESP-KNX-IP-Library_0.4_2894
Se encontraron varias bibliotecas para "EEPROM.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/EEPROM
Se encontraron varias bibliotecas para "WiFi.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 No usado: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Se encontraron varias bibliotecas para "AsyncUDP.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/AsyncUDP
Se encontraron varias bibliotecas para "WebServer.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WebServer
Se encontraron varias bibliotecas para "FS.h"
Usado: /Users/robertson/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

envy, 2020-03-27:
This libary is not designed to work with the ESP32.
AFAIK the ESP32 WiFi functions look different and the library would need to be ported.

closed 2020-03-27 by envy

ascillato, 2020-04-11:
@robetson1989 

This library works as it is also with ESP32.

Now Tasmota ( https://github.com/arendst/Tasmota ) supports also ESP32 and KNX with this library works fine. No modification needed to the library.