envy/esp-knx-ip #83: Port to ESP32?

sauvant, 2020-12-31:
Hi,

I am very interested in this project but it would need to run on ESP32s for me. Do you have any plans to port it? Or has anybody done that already?

Thank and best regards
Keith

c-o-m-m-a-n-d-e-r, 2021-01-11:
Please give it a try : https://github.com/c-o-m-m-a-n-d-e-r/esp-knx-ip

This is just a quick try ... first look was ok for me ... 
will do an PR if everything is fine.

cbovy, 2021-01-26:
@c-o-m-m-a-n-d-e-r For me it works on ESP32 (Lolin D32) with your repo. 
Not sure why 'send_14byte_string' is removed though.

c-o-m-m-a-n-d-e-r, 2021-01-26:
i'm also not sure haha ... will take a look why this was removed.

NilsRo, 2021-03-14:
If everything fine? Have a ESP32 here and like to connect it to my KNX at home to control a dehumidifier, but your change is still not in the master if I am right?

spacegaier, 2021-03-14:
@NilsRo There was never a PR for ESP32 created against this main repo here, so nothing to be merged exists yet.

c-o-m-m-a-n-d-e-r, 2021-03-15:
Yeahr ... totally forgot this ... sorry guys ... will try to check changes and make PR then

mrWheel, 2022-03-20:
Is there a chance this great library will be ported to ESP32 in the near future?

With the latest esp32-core I get tons of compiler errors. It looks like the WiFi lib for the esp32 is not compatible with this library.

jamel-86, 2023-10-04:
I got some errors in the newer versions of Arduino IDE. 
I have forked a new ESP32 port here: https://github.com/jamel-86/ESP32_KNX_IP_Library

hope someone finds it useful 👍

jamel-86, 2023-10-04:
> Is there a chance this great library will be ported to ESP32 in the near future?
> 
> With the latest esp32-core I get tons of compiler errors. It looks like the WiFi lib for the esp32 is not compatible with this library.
> 
> This is the first error I encounter. is there a solution for this error:
> 
> In file included from …/esp-knx-ip.cpp:7:
> esp32-knx-ip.h:231:33: error: flexible array member in union
>     cemi_addi_t additional_info[];
> According to an answer at stackoverflow:
> 
> No, unions do not support flexible array members, only structs. C11 6.7.2.1 §18
> 
> As a special case, the last element of a structure with more than one named member may have an 
> incomplete array type; this is called a flexible array member.
> In addition, zero-length arrays is not valid C, that's a gcc non-standard extension. The reason why 
> you get this to work is because your compiler, gcc, is configured to compile code for the "non-standard 
> GNU language".

Try to change:

` cemi_addi_t additional_info[]; `

with:

` cemi_addi_t` additional_info[i]; `