envy/esp-knx-ip #87: physical address problems for programming in ETS5

richardpub4136, 2021-08-21:
Having solved problems encountered with the https://github.com/arendst/Tasmota version on a Sonoff basic R2,
and as I couldn't get any command results from ETS5 a priori multicast problems, I chose to install your tasmota KNX.ino version of envy / esp-knx-ip.

With this version tasmota KNX.ino esp-knx-ip. , I was able to get a command from the diagnostics mode of ETS5. With this version, with the physical address 1.1.108 the control GA 1/1/2 and the status GA 1/1/3, works.
Unfortunately, since I do not have a participant in TES5 for these GAs, except for the communicator, the Write action in the Diagnostics mode of ETS5 uses the physical address 0.0.1 by default. This is well recognized by the Sonoff Tasmota which turns on and off through the KNX bus? And the state return of the Sonoff Tasmota is done well with the physical address 1.1.108 that I declared in the parameters of the Sonoff Tasmota.
Therefore, the On or Off action, on the switch programmed in ETS5, does not activate the Sonoff Tasmota.

I will try to use the method from https://github.com/thelsing/CreateKnxProd to create an importable file.knxprod participant directly with this app.
But perhaps you would have thanks to your experience another method so that I can associate my 2 AG d in a parameterizable actuator participant directly in ETS5.
Kind regards.

richardpub4136, 2021-08-22:
I made a lot of progress with the transformation of a sonoff into tasmota.
Finally I installed another Tasmota version for my sonoff.ino. which I download with Arduino IDE, and which works correctly and which today allows me to turn my sonoff on and off, using the diagnostic commands in ETS5.
 
On the other hand, I have a problem because I cannot use a switch on my KNX bus network to turn my sonoff on and off because by default my sonoff has a physical participant address of 1.1.108 defined in Tasmota, and a physical participant address. in diagnostics of ETS5 to 0.0.1, and I do not know how to create a virtual participant in ETS5. There are indeed virtual participants in the KNX Association catalog, but I cannot download my GAs there because I don't have a button to integrate this participant into ETS5
(normal, it is virtual. But maybe there is a method)

I found a program on github https://github.com/thelsing/CreateKnxProd
which allows to create a participant file, but this one is in C #
I am therefore obliged to compile after adaptation this program written in C sharp to create a participant file .KnxProd to make an .exe. Unfortunately I don't have C sharp skills

Does anyone have a good command of the C # sharp language because I do not have a good command of this language at all.
At the same time, I am afraid that once I have taken this step, I will also be stuck when downloading in ETS5.

 If anyone knows how to master this language please help me or have any idea on how to force the download
cordially

envy, 2021-08-22:
Please ask questions about other projects on their respective sites or github pages.

>  chose to install your tasmota KNX.ino version of envy / esp-knx-ip.

This project does not provide such a file. I don't where you got it.

closed 2021-08-22 by envy

richardpub4136, 2021-08-22:
sorry i meant:
https://github.com/envy/esp-knx-ip/blob/master/examples/sonoff/sonoff.ino which is an example of esp-knx-ip. It works perfectly and esp-knx-ip will disturb me to directly retrieve the data from my weather station running on an Arduino Mega, in my KNX bus.
the problem I am having is that ETS5 does send its commands to the Sonoff but that I cannot control it by a switch installed in my KNX project with the GA which activates the sonoff because I cannot install a virtual participant.
sorry if i missed any specifics.

envy, 2021-08-22:
So you can receive data from your esp-knx device but cannot send data to it?

Either you:
a) Modify the filter table of your IP router to allow the specific group addresses to be send to the network (https://support.knx.org/hc/en-us/articles/360007445460-Couplers-Filter-Tables) or
b) Use a KNX "dummy application"/"dummy device" inside your IP area to represent the esp.knx device (https://support.knx.org/hc/en-us/articles/360008345760-Dummy-devices)

richardpub4136, 2021-08-22:
Knowing that I can write the sonoff on or off, by writing these values to the sonoff GA using the diagnostics mode of ETS5, I assume that the transfer of these data through the KNX bus is working. But I do not understand the routing mode between the KNX bus and the sonoff. Besides, this one sends me back its status through its GA to ETS5.
My concern is rather to be able to configure an AG in a virtual participant whose physical address I determine identical to that of the sonoff in my case 1.1.108.
I am looking towards virtual KNX to fix this address, but I have a doubt about the possibility of running Virtual KNX at the same time as a real KNX project ???
I will test....
Don't hesitate to tell me if you think I'm completely wrong ...

richardpub4136, 2021-09-28:
is your version compatible with programming by ETS5? or do you have to create a .knxprod file to program the Sonoff ??
where should I turn to versions compatible with ETS5 to flash the ESP8266 ???

Cordiallement

envy, 2021-09-28:
No, this is not programmed by the ETS.

If you want this, you need to look at https://github.com/thelsing/knx